multiwii/baseflight #245: Not an issue but feature request

ecmnet, 2015-02-13:
In my current project, I use baseflight as low-level flight controller. However, for high-level control I use a linux-based platform to do complex math (extended Kalman, etc). Communication to baseflight is via the serial interface. Currently I've modified baseflight to accept a new command, which is RC_CORRECTION to add offsets to the remote control calculated by the high-level controller. It would be nice to implement this command to base flight-master. Or, any other ideas to communicate to baseflight, because serial communication has quite a big latency? If interested, I can share the modification.

ecmnet, 2015-02-13:
btw. I am using GPS via the high level controller as well as SFA10 for height control and PX4Flow for real.position control...

EmilsPa, 2015-09-15:
Make a PPM output and emulate a normal RC receiver.
No latency and you can use any Flight controller as low level controller.

closed 2016-03-30 by ecmnet