multiwii/baseflight #246: Servo output no longer re-centers itself to mid after changing rate % in servo tab

creyc, 2015-02-16:
This issue can be demonstrated in the 2015.02.12 firmware by setting the airplane mixer then changing any servo to 'Rate: -100%' in the servo tab then checking that servo's output in the Motor Testing tab.  Expected output is MID setting, but instead getting MIN.

This issue first shows up in the Dev 2014.12.08 build, but was working correctly in Stable 2014.12.04 build.

ShikOfTheRa, 2015-03-09:
I am seeing the same issue

fiendie, 2015-03-09:
@schugabe is currently working on MSP for the new servo mixer. As soon as that's done the servo tab can be fixed by @ctn-dev

schugabe, 2015-03-21:
Can someone test the fix I just pushed?

cTn-dev, 2015-03-24:
Any word on this one ?

creyc, 2015-03-25:
@schugabe rate values < 0% seem to work correctly now in my testing, thanks!

schugabe, 2015-03-27:
Great! Can someone close this issue?

ShikOfTheRa, 2015-03-27:
Seems to work for me too.. Muchos Gracias

fiendie, 2015-03-27:
@schugabe thanks a lot ;)

closed 2015-03-27 by fiendie