multiwii/baseflight #249: Tricopter 3d Mode

smarblewnec, 2015-02-24:
Hello,
There is an issue with tricopters in 3d mode.  The only function missing is that the rudder servo needs to reverse when the throttle goes negative.  This is because the thrust angle on the rudder motor changes 180 degrees off from what the flight controller expects when the motors reverse.  Currently, once a tricopter is inverted it spins uncontrollably.  
Thank you for your help!

catchthewind, 2015-03-03:
Please fix it. It is impossible to use tricopter inverted in 3d mode :(

trollcop, 2015-03-04:
You're welcome to submit a patch, it is trivial.

catchthewind, 2015-03-04:
My coding experience is not good enough to risk doing it :( 
However I would more than happy to paypal you a beer for doing this trivial patch ;)

trollcop, 2015-03-04:
I think someone is on it already.

fiendie, 2015-03-04:
@catchthewind: Could you try this? http://firmware.baseflight.net/baseflight_NAZE_2015-03-04_09-43-00.hex
This is not flight tested obviously so maybe do a dry-run without props first? ;)

catchthewind, 2015-03-04:
I wont be home till friday to test it :(
@smarblewnec could you test it faster and post the result?

fiendie, 2015-03-09:
It's Monday now, anything to report? ;)

catchthewind, 2015-03-09:
I have bench tested it ok but I have also broken my trifecta frame as well as one engine and waiting for HK with backorder :( I wont be able to flight test it for a few weeks cause order is waiting for babybeast motor to be restocked :(

@smarblewnec can You flight test it in the meantime?

smarblewnec, 2015-03-10:
Hey guys, sorry I missed the progression on this.  I can test it out within the next couple of days.  I am also waiting for a part to arrive for my tricopter after I broke it this weekend.