multiwii/baseflight #262: GPS information seemingly incorrect

dustin, 2015-03-20:
I'm not entirely sure which bits are MWOSD and which are baseflight, but I was testing out GPS last night and saw a couple things that are wrong.

In the picture below, I'm standing behind the right side of the artificial horizon vertical bars.  The direction home arrow is pointing to the left.  The distance from home might be off slightly, but has been close enough.

The altitude reset at some point and started reporting negative numbers.

Max altitude is reported at 0 after my flight.

[image]

DTFUHF, 2015-03-20:
Can you verify the heading is correct? Or is it wrong? If so, by how much?

In MWOSD, did you un-comment #define BASEFLIGHT ?

dustin, 2015-03-20:
Yes, I did `#define BASEFLIGHT`.  Note that this is naze32 acro.  It's getting heading from somewhere, but I don't fully understand where.

I'm not entirely sure which heading that's showing there.  It looks like it's showing heading home, which isn't useful here.  :/  It seemed to drift around during flight.  I'll add compass to my OSD and do a bit more research.

ShikOfTheRa, 2015-04-09:
This should be closed - at current time MWOSD doesn't have support for correct acro heading without compass... next few weeks....

dustin, 2015-04-09:
Are you implying MWOSD is going to have a fix?  That'd be pretty awesome.

closed 2015-04-09 by dustin

fiendie, 2015-04-09:
@dustin He should know ;D