multiwii/baseflight #269: Baro may return wrong value

ledvinap, 2015-04-03:
This bug is present (but probably much harder to trigger) in baseflight: https://github.com/cleanflight/cleanflight/pull/720

trollcop, 2015-04-03:
> cleanflight/cleanflight#720

unless you're actually contributing something (like working code), may I
politely request you to go fuck yourself, good sir?

digitalentity, 2015-04-03:
@trollcop, actually @ledvinap got a point. This bug is likely present in baseflight. Took a while to figure it out here cleanflight/cleanflight#569.
In baseflight this line 
https://github.com/multiwii/baseflight/blob/5fffabbc379dd4cc684fcc2f8b2e940279be7d00/src/sensors.c#L361
probably should be changed to 
`baroDeadline = micros();`
to fix this.

trollcop, 2015-04-03:
great, go ahead and change it.
I'm not clicking any links with joblessflight in them.