multiwii/baseflight #271: motor PWM

DietmarM, 2015-05-03:
Using GUI 0.57 and baseflight from 01.05.2015 I had this issue:

motor_pwm_rate = 0 - could not set to a good value - gui said 1000..2000 valid only ?!?
(Strange vibrations with this value - because much to slow)

went back to older Version and copter works fine.

readerror67, 2015-05-03:
Current release is 0.63, 0.57 is quite old.

DietmarM, 2015-05-03:
OK - but this is the latest you can download on github.com

readerror67, 2015-05-03:
https://chrome.google.com/webstore/detail/baseflight-configurator/mppkgnedeapfejgfimkdoninnofofigk?hl=en

DietmarM, 2015-05-03:
OK - now with the new gui and a new flashed Firmware it works - but Motor PWM rate seems to be nailed to 400 - cant Change it to 480 any more

readerror67, 2015-05-03:
Entering CLI Mode, type 'exit' to return, or 'help'

# set motor_pwm_rate
motor_pwm_rate = 400

# set motor_pwm_rate=480
motor_pwm_rate set to 480
# save
Saving...
Rebooting...
Entering CLI Mode, type 'exit' to return, or 'help'

# set motor_pwm_rate
motor_pwm_rate = 480

# version
Afro32 CLI version 2.3 Mar 27 2015 / 10:52:12

DietmarM, 2015-05-03:
OK - now I was able to reproduce it: there is a selection you can make at Configuration tab: brushed motors. This was empty first - able to Change PWM - now I changed it to off - 400 fixed. 
...A bit strange: reboot and I was able to Change it and brushed got empty again - can't find the logic so far.

trollcop, 2015-05-03:
PWM rate higher than 500 (?I think?) automatically switches to brushed mode.

DietmarM, 2015-05-03:
no above 500 is not possible by physics - but more than 400. It does not take values above 400 - but I can't tell how it happens.

DietmarM, 2015-05-03:
I have to leave now - will try later again why and when it is strange.
It would be nice to see the PWM frequency in the gui when it is below 500 - or to be able to Change it there. Putting to off sets 400.

trollcop, 2015-05-03:
if (pwm_params.motorPwmRate > 500)
        pwm_params.idlePulse = 0; // brushed motors

btw, tehres no reason to hve it above 400 anyway.

DietmarM, 2015-05-03:
Faster controle Loop is always a reason - the kiss takes it up to the physical limmit as far I know.
But I was staying below 500 always (< not <=).

luggi, 2015-05-03:
@DietmarM there is also sync_pwm and==oneshot fast_pwm==125µs pulses as feature options now..

DietmarM, 2015-05-03:
now I changed the pitch trim in configuration - and its dangerous: Motor pwm got changed back to 0 again - it takes 1 sec to Switch off a crazys quad.

DietmarM, 2015-05-03:
There is also a report about the bug in a german forum: 

http://www.kopterforum.de/topic/30718-info-baseflight-gui-l%C3%B6scht-motor-pwm-rate/

When the PWM rate is <> 400 and save is pressed in configuration tab - pwm rate gets set to 0 

Afterwards this is dificult to get Bach to a 400 or other value.

trollcop, 2015-05-03:
I think you need to open a bug @ configurator and stop spamming here - I
don't think this is baseflight related at all

closed 2015-05-03 by DietmarM