multiwii/baseflight #272: FastPWM timer re-enable

gke, 2015-05-18:
Should the timer re-enable in pwmWriteSyncPwm be delayed until all channels for that timer, or better all timer channels are set?

trollcop, 2015-05-18:
I don't know, what do you think?

gke, 2015-05-18:
Not absolutely sure but the issue relates to Fiendie's branch where I think the one-shot code is currently. I see anomalies in my logic analyser trace and these seem to go away when the the re-enable is done outside the motor setpoint update loop that calls pwmWriteSyncPwm. I just re-enable all of the PWM timers at once.

trollcop, 2015-05-18:
pretty sure cumshot stuff is in main branch now, but I'd be surprised if
anyone uses or cares about it.

gke, 2015-05-18:
I am definitely NOT a oneshot believer, trust me, just curious. All seems like solid gold wiring on sound system bull.  My UAVX code just runs async and at around 500Hz it is good enough ;). I think I do believe in spin up and spin down symmetry though.

closed 2015-05-18 by gke

trollcop, 2015-05-18:
I'm pretty sure without external triggering there's no way to guarantee
separate timers to start at same time.

But I've not looked into it for all the reasons you've already stated.

fiendie, 2015-05-18:
@gke, the stuff was merged long ago. There is some real data about the "oneshot125" stuff. The gist of it is that the "oneshot" part (SyncPWM) actually yields some improvements but the "125" part, namely FastPWM, is questionable at best. Which is why we split it up into two features in the first place. SyncPWM also works with any ESC. A little more than "anomalies" would also be pretty helpful to narrow down if there really is a problem :)