multiwii/baseflight #277: mAhdrawn using [Ah] resolution, so reporting 0 in most of the flights

ciafe, 2015-06-10:
Hello,

I'm using a NAZE32_REV5 with FEATURE_VBAT ON and power_adc_channel = 5 with a ACS756 sensor connecte to it.

The VBat and amperage are correcly measured and transmitted, but the mAhdrawn remains always 0 because the resolution is 1Ah, so it will take time, or never in one filght, to get a value different than 0. Based on this code:

                mAhdrawnRaw += (amperage * vbatCycleTime) / 1000;
                mAhdrawn = mAhdrawnRaw / (3600 * 100);

Is my understanding corrent?
If yes, does it make sense to work with [mAh] resolution instead of [Ah]?

Thanks a lot!

luggi, 2015-06-10:
You must be doing something else wrong, for me it's working perfectly at mAh resolution.
Next time please post a link to the code in the git repository.

readerror67, 2015-06-10:
Well they are integers so using Ah doesn't make much sense, I've been using current sensing for a while with MWOSD and not had any issues.

luggi, 2015-06-10:
btw: https://github.com/multiwii/baseflight/blob/ffde4a3d57f4616be85592f0e7b9b16a778306d6/src/mw.h#L472 ;)

ciafe, 2015-06-11:
OK, thanks for you feedback, I'll check my setup again!

And sorry for bothering you, there is a clear comment in the variable definition:
extern int32_t mAhdrawn;              // milli ampere hours drawn from battery since start