multiwii/baseflight #296: Flywing Servo mising not working on latest release

Nandox7, 2015-09-06:
Hey,

Tried to update to the latest FW release (2015-08-27) and the servo mixing in Fling Wing Mode doesn't work.
Reverting to the previous one, makes it work. (2015-07-11)

Checking latest commits this one may be the culprit: ba9c0d296ce65ce9b1e48c325a9355b16a9006d1
I'll be running a few more tests with it.

trollcop, 2015-09-06:
I'll be looking forward a patch that fixes it ;)

EmilsPa, 2015-09-08:
Non of the Servo models work any more.
Planes and TRI copter
The servo Tab don't contain any data.

trollcop, 2015-09-08:
cool

readerror67, 2015-09-08:
Can't we just roll back the stuff lazd commited ?

trollcop, 2015-09-08:
how about getting him to fix it instead?

readerror67, 2015-09-08:
Hes over on cleanflight now?

Nandox7, 2015-09-08:
I'm on Holiday's shouldn't be looking into code but...I want it to fly my
wing. :)

Yesterday I checked it a bit. Seems the mixer is a bit messed.
I got it just half working by changing the custom servo mixer to load only
when the new feature is enabled.
I'd get elevator mix but not ailerons so there is more to check in there.

I could be wrong but seems the mixer rules are overriding itself.

schugabe, 2015-09-08:
https://github.com/multiwii/baseflight/blob/ba9c0d296ce65ce9b1e48c325a9355b16a9006d1/src/mixer.c#L333 
This condition in this line is wrong!
It should be mcfg.mixerConfiguration == MULTITYPE_CUSTOM_PLANE

schugabe, 2015-09-08:
Actually it's not that easy any more. It should load the custom mixer if we have a MULTITYPE_CUSTOM_PLANE or the feature SERVO_MIXER is active and the current model type does not have a default servo mixer.

Or should the feature SERVO_MIXER always activate the custom servo mixer chain?

```c
if (mcfg.mixerConfiguration == MULTITYPE_CUSTOM_PLANE || feature(FEATURE_SERVO_MIXER))
            loadCustomServoMixer();
```

Nandox7, 2015-09-08:
@schugabe that was my first change as well. But doing that only makes the elevator mix work the ailerons still don't work.

What I did was to leave the current check and add and extra dor the feature.

if (core.useServo & feature(FEATURE_SERVO_MIXER))
         loadCustomServoMixer();

dustin, 2015-09-09:
@lazd Any input?  Your change seems to have negatively impacted people.

trollcop, 2015-09-09:
left a turd at front door and ran away?

lazd, 2015-09-09:
I don't see why it wouldn't work, as it checked `useServo` for the mixer type and ran the servo mixer if it was on. Maybe @schugabe and @Nandox7 were unable to setup their servo mixes correctly using the `smix` command.

lazd, 2015-09-09:
> https://github.com/multiwii/baseflight/blob/ba9c0d296ce65ce9b1e48c325a9355b16a9006d1/src/mixer.c#L333 
> The condition in this line is wrong!
> It should be mcfg.mixerConfiguration == MULTITYPE_CUSTOM_PLANE

No, that would defeat the purpose of the pull request, titled "Make servoMixer work for types other than CUSTOM_PLANE." As `MULTITYPE_CUSTOM_PLANE` defines `useServo = 1`, we still use the mixer for that multitype.

lazd, 2015-09-09:
@schugabe @Nandox7 with the release you were having problems with previously, try setting up your servo mixer rules differently. The mapping for `MULTITYPE_FLYING_WING` seems to be target channel 3 -> servo 0  and target channel 4 -> servo 1.

```
# Control servo output 3 with channel 1 (Aileron)
smix 1 3 1 100 0 0 100 0

# Control servo output 4 with channel 2 (Elevator)
smix 2 4 2 100 0 0 100 0
```

Are you also running a gimbal by chance?

lazd, 2015-09-09:
@Nandox7 @schugabe please post a gist with the output of `dump`. Without seeing your features you have enabled or servo mixes, I'm troubleshooting this completely in the dark.

lazd, 2015-09-09:
> I could be wrong but seems the mixer rules are overriding itself.

@schugabe that might be the case. Given that we have a number of different ways to setup servos, it seems my assumption that the servo mixer could run for any multitype that uses servos was wrong -- mutlitypes have their own, hardcoded servo mixer rules that are getting blown away.

Ideally, when a given multitype is selected, the default rules would be added to the custom servo mixer for that type. Instead, we hardcode them completely separate from the custom servomixer, hence the situation my PR got us into.

Nandox7, 2015-09-09:
@lazd I thought so.
Because I never had to set any mix when using the Flying Wing type. It makes the mix automatically and I just have to reverse the servo movements in the UI in case one of the channels is reversed (aileron/elevator).

Not sure what's the ideal solution here, but moving into a pure smix style could be one.
When setting a mode it would automatically insert the rules using smix.
This will break the UI i believe?

schugabe, 2015-09-10:
The way I implemented the servo mixing had a reason: clean up the hard coded mess. Custom servo mixing was more or less a nice by product because the implementation was flexible enough to support it as well. 

My main problem with the suggested "load default rules to smix storage in all cases" is that there are no API/events that would allow to implement this without touching several different places where the actual model is set. Many different use cases must be considered. It results in a more complex implementation without any real benefit.

I think that most of the people just want to select a model and are done => therefore the load default rules based on model type during initialization is the default case and imho should stay the default case.

Just out of curiosity: What models do need the feature servo mixer?

lazd, 2015-09-10:
@schugabe I wanted to use the servo mixer to map an AUX switch to a servo output for a QUADX. This is insanely easy to do on the CC3D with OpenPilot, and I was surprised that it was impossible when I moved to Naze32 + Baseflight. Baseflight is full of hardcoded features, such as support for gimbals that sits behind a feature, enabling servos for QUADX, but giving zero flexibility when it comes to configuring or overriding anything. I added the SERVO_MIXER feature in hopes of making the flexibility of the servo mixer available everywhere, but it seems there are just too many hardcoded features we have to work around for this to be both flexible and easy.

Nandox7, 2015-11-03:
I was away for some time but picking on this again.
Is this dead? I haven't seen any commit related to it, so I'd imagine the issue is still here?

trollcop, 2015-11-03:
Should be still there, I haven't seen anyone doing anything with this.
feel free to test/fix/pr :)

lazd, 2015-11-03:
@Nandox7 the commits were rolled back, so the issue is fixed. https://github.com/multiwii/baseflight/commit/436f9264ed9147fc5cd0272638ea0cd27af9ffd4

That said, it's my opinion that a **unified** way of handling servo mixing that removes the hardcoded features and brings everything together under `smix` would be a great step forward.

trollcop, 2015-11-04:
> the issue is fixed. 436f926

patches welcome, especially if they're not hacks

tracernz, 2015-11-04:
@lazd rolling the commits back doesn't mean it's not welcome, just that it needs a little rework to iron out the issues. ;-)

EmilsPa, 2015-11-04:
Could you maybe roll back on the Hex file to?
The current stable in Gui don't work on any models with servos!..