multiwii/baseflight #297: Motors twitch when disarmed with fastpwm+syncpwm

dustin, 2015-09-09:
I've noticed that when I've got power, but am disarmed, my motors will kind of twitch while disarmed.  They'll occasionally turn a bit in short bursts.

I turned off fastpwm and it stopped twitching.

trollcop, 2015-09-09:
upgrade your esc firmware and disable cumshot

dustin, 2015-09-17:
The motors do not twitch in the exact same config with cleanflight (which brought enough of its own bugs to not last long on this build).

I'm not sure what I can do to help, but it seems there's a bug somewhere that's causing pulses to be inconsistent enough to cause motors to try to turn when disarmed.

trollcop, 2015-09-18:
Patches welcome

dustin, 2015-09-18:
Example capture from motor 1 (I see it in all of them, but hope this is somewhat helpful):

```csv
Time [s], Analyzer Name, Decoded Protocol Result
0.002472416666667,PWM,127 μS
0.082563750000000,PWM,134 μS (+7)
0.085076333333333,PWM,127 μS (-8)
0.192778333333333,PWM,134 μS (+7)
0.195235500000000,PWM,127 μS (-7)
0.388101166666667,PWM,134 μS (+7)
0.390556000000000,PWM,126 μS (-8)
0.558317583333333,PWM,134 μS (+8)
0.560827750000000,PWM,127 μS (-7)
0.613380166666667,PWM,132 μS (+6)
0.615894666666667,PWM,127 μS (-6)
0.716033916666667,PWM,134 μS (+7)
0.718550916666667,PWM,127 μS (-7)
0.873751583333333,PWM,134 μS (+8)
0.876261333333333,PWM,126 μS (-8)
0.966359500000000,PWM,134 μS (+8)
0.968930500000000,PWM,127 μS (-7)
1.492115500000000,PWM,135 μS (+8)
1.497236500000000,PWM,126 μS (-8)
1.562225000000000,PWM,134 μS (+8)
1.564788500000000,PWM,126 μS (-8)
1.772547500000000,PWM,134 μS (+8)
1.775105750000000,PWM,127 μS (-8)
1.850156250000000,PWM,134 μS (+8)
1.852655166666667,PWM,126 μS (-8)
2.203195583333333,PWM,134 μS (+8)
2.205757083333333,PWM,127 μS (-8)
2.353411916666666,PWM,134 μS (+7)
2.355970500000000,PWM,127 μS (-7)
2.413519416666666,PWM,134 μS (+7)
2.416077333333333,PWM,127 μS (-7)
2.418518916666667,PWM,134 μS (+7)
2.421018666666667,PWM,127 μS (-7)
2.706451583333333,PWM,134 μS (+8)
2.708950500000000,PWM,127 μS (-8)
2.876773083333333,PWM,134 μS (+8)
2.879217250000000,PWM,127 μS (-8)
2.914274000000000,PWM,134 μS (+7)
2.916785333333333,PWM,127 μS (-7)
2.986882916666667,PWM,134 μS (+7)
2.989375666666667,PWM,127 μS (-7)
3.214706500000000,PWM,134 μS (+7)
3.217269166666667,PWM,127 μS (-8)
3.272313583333333,PWM,134 μS (+8)
3.274827083333333,PWM,127 μS (-8)
3.307419500000000,PWM,134 μS (+8)
3.309869750000000,PWM,127 μS (-7)
3.349923166666667,PWM,134 μS (+7)
3.352496000000000,PWM,126 μS (-8)
3.382528000000000,PWM,134 μS (+8)
3.384976166666667,PWM,127 μS (-8)
3.717963416666667,PWM,134 μS (+8)
3.720476416666667,PWM,127 μS (-7)
3.753069333333333,PWM,134 μS (+8)
3.755523500000000,PWM,127 μS (-8)
3.855677500000000,PWM,134 μS (+7)
3.858239666666667,PWM,127 μS (-7)
4.048501500000000,PWM,134 μS (+7)
4.050993416666667,PWM,127 μS (-8)
4.126109166666667,PWM,134 μS (+8)
4.128632083333334,PWM,127 μS (-8)
4.228719333333333,PWM,134 μS (+7)
4.231207250000000,PWM,127 μS (-7)
4.298826750000000,PWM,134 μS (+7)
4.301354333333333,PWM,127 μS (-7)
4.381435166666667,PWM,134 μS (+8)
4.383950750000000,PWM,127 μS (-8)
4.386435166666667,PWM,134 μS (+8)
4.388961166666666,PWM,127 μS (-8)
4.526650833333333,PWM,134 μS (+8)
4.529211500000000,PWM,126 μS (-8)
4.626864666666667,PWM,134 μS (+8)
4.629331500000000,PWM,127 μS (-8)
4.646866000000000,PWM,134 μS (+8)
4.649426166666666,PWM,127 μS (-8)
```

dustin, 2015-09-18:
Yeah, I'd say this looks right:

[image]

@readerror67 suggested lowering the mincommand to go above the jitter there.

dustin, 2015-09-18:
I was also able to reproduce this with a different board on the desk with logic analyzer hooked up to outputs and the following config:

```
aux 0 0
aux 1 0
aux 2 2
aux 3 0
aux 4 0
aux 5 0
aux 6 0
aux 7 0
aux 8 0
aux 9 0
aux 10 0
aux 11 0
aux 12 0
aux 13 128
aux 14 0
aux 15 0
aux 16 0
aux 17 0
aux 18 0
aux 19 0
aux 20 0
aux 21 0
aux 22 0
aux 23 0
mixer QUADX
servo 1 1020 1500 2000 30
servo 2 1020 1500 2000 30
servo 3 1020 1500 2000 100
servo 4 1020 1500 2000 100
servo 5 1020 1500 2000 100
servo 6 1020 1500 2000 100
servo 7 1020 1500 2000 100
servo 8 1020 1500 2000 100
feature -PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -SERIALRX
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -LED_RING
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -POWERMETER
feature -VARIO
feature -3D
feature -FW_FAILSAFE_RTH
feature -SYNCPWM
feature -FASTPWM
feature -SERVO_MIXER
feature PPM
feature VBAT
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature POWERMETER
feature SYNCPWM
feature FASTPWM
map AETR1234
set looptime = 2500
set emf_avoidance = 0
set midrc = 1500
set minthrottle = 1100
set maxthrottle = 1850
set mincommand = 1000
set mincheck = 1100
set maxcheck = 1900
set deadband3d_low = 1406
set deadband3d_high = 1514
set neutral3d = 1460
set deadband3d_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set pwm_filter = 0
set retarded_arm = 0
set disarm_kill_switch = 1
set fw_althold_dir = 1
set reboot_character = 82
set serial_baudrate = 115200
set softserial_baudrate = 9600
set softserial_1_inverted = 1
set softserial_2_inverted = 0
set gps_type = 0
set gps_baudrate = 0
set gps_ubx_sbas = 0
set gps_autobaud = 0
set serialrx_type = 0
set spektrum_sat_bind = 0
set spektrum_sat_on_flexport = 0
set telemetry_provider = 0
set telemetry_port = 1
set telemetry_switch = 0
set vbatscale = 110
set currentscale = 400
set currentoffset = 0
set multiwiicurrentoutput = 0
set vbatmaxcellvoltage = 43
set vbatmincellvoltage = 32
set vbatwarningcellvoltage = 29
set power_adc_channel = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set yaw_control_direction = 1
set acc_hardware = 0
set mag_hardware = 0
set max_angle_inclination = 500
set moron_threshold = 32
set gyro_lpf = 42
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set pid_controller = 0
set deadband = 0
set yawdeadband = 0
set alt_hold_throttle_neutral = 40
set alt_hold_fast_change = 1
set throttle_correction_value = 0
set throttle_correction_angle = 800
set rc_rate = 100
set rc_expo = 50
set thr_mid = 50
set thr_expo = 0
set roll_rate = 66
set pitch_rate = 66
set yaw_rate = 70
set tpa_rate = 31
set tpa_breakpoint = 1500
set failsafe_delay = 10
set failsafe_off_delay = 5
set failsafe_throttle = 1200
set failsafe_detect_threshold = 985
set auto_disarm_board = 5
set rssi_aux_channel = 0
set rssi_adc_channel = 0
set rssi_adc_max = 4095
set rssi_adc_offset = 0
set yaw_direction = 1
set tri_unarmed_servo = 1
set gimbal_flags = 1
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set acc_unarmedcal = 1
set small_angle = 25
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set accz_lpf_cutoff =  5.000
set mag_declination = 0
set gps_pos_p = 11
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 20
set gps_posr_i = 8
set gps_posr_d = 45
set gps_nav_p = 14
set gps_nav_i = 20
set gps_nav_d = 80
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set p_pitch = 32
set i_pitch = 42
set d_pitch = 23
set p_roll = 32
set i_roll = 32
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set fw_gps_maxcorr = 20
set fw_gps_rudder = 15
set fw_gps_maxclimb = 15
set fw_gps_maxdive = 15
set fw_climb_throttle = 1900
set fw_cruise_throttle = 1500
set fw_idle_throttle = 1300
set fw_scaler_throttle = 8
set fw_roll_comp =  1.000
set fw_rth_alt = 0
```

tracernz, 2015-09-18:
I also get this +/- 7-8us jitter on the bench when SyncPWM/Oneshot is enabled (with or without FastPWM). The jitter is much more frequent when Softserial is enabled.

dustin, 2015-09-19:
I get no jitter without softserial.  Seems to be messing with stuff.

quadquestions, 2015-10-14:
try to lower your mincommand to 950--it should stop the jitters.

dustin, 2015-10-14:
It does, but that's just because the jitter is below motor activation speed.  Having zero jitter is better.  lux had some ideas about how to clean up timing with softserial enabled.

closed 2021-03-31 by dustin