multiwii/baseflight #302: Singlecopter 1M Channel Problem

ChrisKrieg, 2015-12-22:
Servo signal is coming through on pins 1 and 2. Pin 6 already burned up one of my servos. It's as if the signals are mapped using the quad with servo gimbal. Servos on 1 and 2 with motor signals on 4 through 6. This issue pretty much eliminates the possibility of using the singlecopter profile with Baseflight until it is solved.
KioLiex posted about this issue in December of last year in #231 but there seems to have been no progress.

Can someone please fix this glitch? I wish I had the coding skills to do it myself, but I must humbly rely on the expertise of others.

trollcop, 2015-12-23:
if you enable gimbal there are no servo outputs left unless you use rc pins
as servo and use ppm.

I'm not even sure what the question is

ChrisKrieg, 2016-02-19:
Sorry for the huge delay. I never got an email from GitHub for  your reply.

My issue is that I have not enabled the gimbal. I don't want to use one. I just want to get my 4 servos working correctly but as I described, the signals coming out of the pins are not correct (at least on my board). I am reading servo signals from pins 1 and 2, but the other pins are not sending the proper signals that the configuration states that they should. I'll try uploading the config again this weekend and see if there is any change.