multiwii/baseflight #313: Altitude and velocity values

NathanNoumowe, 2017-07-13:
Hello, I am trying to trigger a parachute with a naze32 Acro and for that I need to get the altitude and velocity values. So I used printf to watch their values:

```
printf("Alt = %d ",(int)accAlt);
printf("VitZ = %d\n",( int)accZ_tmp);
```

But the altitude value isn't right at all. It's changing pretty slowly and the values are huge while it's suppose to be in centimeters. The velocity have pretty much same problems.
The values get also completly crazy when I turn the card upside down.
Is it normal? If it's not how can I fix this?

Thank you in advance for your answers.

tracernz, 2017-07-13:
Uhhh, acro variant doesn't have baro right?

NathanNoumowe, 2017-07-13:
Yes it does have one.

NathanNoumowe, 2017-07-18:
Anybody?

trollcop, 2017-07-22:
There's nothing to fix, look at msp code or telemetry, altitude value is
there.
What you need to do is zero it at takeoff (I believe there's a key combo).
Then the numbers will be relative to ground.

NathanNoumowe, 2017-07-24:
Okay thanks I'm gonna try that. Otherwise I fixed the problem with the velocity when I turn the card upside down. Files that i have were already edited by someone before me and there was a delay in the main. When I took it off, values stopped getting bigger and bigger when I flip the card. But for now the altitude value is still wrong.

NathanNoumowe, 2017-07-24:
I found this at the line 130, in the file telemetry_hott.c :
```
// Altitude
HoTTV4GPSModule.altitudeLow = GPS_altitude & 0x00FF;
HoTTV4GPSModule.altitudeHigh = GPS_altitude >> 8;
```
I'm not sure if it contain altitude value but I can not display it.

trollcop, 2017-07-24:
thats *GPS* altitude, you want the one from BARO, i already explained where
it was.
If you have no GPS then clearly there's not gonna be valid value in there.