multiwii/baseflight #316: Baseflight will flash flight controller but cannot connect.

aescobar96, 2020-11-28:
Hello, I am making a flight controller using a bluepill and interfacing with an FTDI board. In the past I have been able to flash and connect the board using baseflight. Now It seems I can only flash the board but I cannot connect after I have flashed it. Here is what I get when I attempt to connect to it:

15:34:37 -- Serial port successfully opened with ID: 36

15:34:47 -- No configuration received within 10 seconds, communication failed

15:34:48 -- Serial port successfully closed

Any insight would help. I have been taking the jumpers off the boot pins after flashing

wisess, 2022-11-07:
Hello! I have same error with my blackpill board. I used stm32f401ccu6 controller. I try to flash the board in serial mode and with st-link programmer but also I have an error.

wangjun000, 2022-11-07:
您好，邮件已经收到！！！！！              这是来自QQ邮箱的自动回复邮件。

trollcop, 2022-11-07:
> Hello! I have same error with my blackpill board. I used stm32f401ccu6 controller. I try to flash the board in serial mode and with st-link programmer but also I have an error.

baseflight never supported anything other than STM32F103. Whatever you are running is not baseflight.