multiwii/baseflight #317: Altitude mode not working as it should..

Lalbabu2001, 2023-08-24:
Hey, friends 
I flashed baseflight on my naze 32 bmp280 
Problem is quad not using midvalue 1500 as its altitude mode as netrual position.
It's using the min throttle value in case of power on the quad>> switch into baro mode>> arm the quad and lift off.
And the value of throttle when you switch into altitude hold mode.
Please help , i want my quad to maintain altitude at 1500 throttle value.

wangjun000, 2023-08-24:
您好，邮件已经收到！！！！！              这是来自QQ邮箱的自动回复邮件。

Lalbabu2001, 2023-08-24:
So please help

trollcop, 2023-08-24:
It's extremely unlikely you're running baseflight or on my hardware, so I
have no idea what the problem is.

Lalbabu2001, 2023-08-25:
Till date it works great with bare minimum hardware,
Listen what actually happens is the quad in baro mode starts to climb as you raise throttle just as work in acro or level mode.
What it should do..
It should not respond until mid throttle in baro mode, it should should respond when the throttle is raised abouve mid value..
I think the quad is using initial throttle position in baro mode. I want it use mid throttle value. Like arducopter does.